hjd1964/OnStepX #2: Errors when trying to compile with GPS

kbahey, 2021-07-17:
When I try to compile with GPS (for the S6), I get the following errors 

```
sketch/src/telescope/mount/site/Site.cpp: In function 'void gpsCheck()':
sketch/src/telescope/mount/site/Site.cpp:24:40: error: 'elevation' was not declared in this scope
   24 |       tls.getSite(latitude, longitude, elevation);
      |                                        ^~~~~~~~~
sketch/src/telescope/mount/site/Site.cpp:34:7: error: 'dateIsReady' was not declared in this scope
   34 |       dateIsReady = true;
      |       ^~~~~~~~~~~
sketch/src/telescope/mount/site/Site.cpp:35:7: error: 'timeIsReady' was not declared in this scope
   35 |       timeIsReady = true;
      |       ^~~~~~~~~~~
exit status 1
```

kbahey, 2021-07-21:
Simple fix for the elevation part, just declare the variable:

```
diff --git a/src/telescope/mount/site/Site.cpp b/src/telescope/mount/site/Site.cpp
--- a/src/telescope/mount/site/Site.cpp
+++ b/src/telescope/mount/site/Site.cpp
@@ -21,6 +21,7 @@ IRAM_ATTR void clockTickWrapper() { centisecondLAST++; }
     if (tls.ready) {
       VF("MSG: Tls_GPS, setting site from GPS");
       double latitude, longitude;
+      float elevation;
       tls.getSite(latitude, longitude, elevation);
       site.location.latitude = degToRad(latitude);
       site.location.longitude = degToRad(longitude);
```

The other two errors are more involved, since gpsCheck is not a class member method. Trying to make it so causes other errors when it is called in tasks.add()

hjd1964, 2021-07-21:
I fixed these

closed 2021-07-21 by hjd1964

kbahey, 2021-07-21:
Thanks. 

Now a different error:

sketch/src/telescope/mount/Mount.h:176:5: error: 'Axis' does not name a type
  176 |     Axis axis1;
      |     ^~~~
sketch/src/telescope/mount/Mount.h:177:5: error: 'Axis' does not name a type
  177 |     Axis axis2;

hjd1964, 2021-07-22:
I missed merging in a bit of code and that was tripping it up vs. what I
have here.

Should be fixed now.  The software serial GPS should be ok now too.

S6 compiles cleanly for me with GPS on.

hjd1964, 2021-07-22:
The additional settings for GPS...

#define SERIAL_GPS             SoftSerial
#define SERIAL_GPS_BAUD              4800

Not needed but supported (automatically assigned in the S6 HAL):
#define SERIAL_GPS_RX                 PA3
#define SERIAL_GPS_TX                 PA2

I gave it a generic name "SoftSerial" as this facility can be used for
other boards.

I didn't provide for a default baud rate since it is my understanding most
are either 4800 or 9600 so *perhaps*? better to just throw an error and
make the user specify it.

hjd1964, 2021-07-22:
Unrelated, but I'd like to hear your opinion about an aspect of OnStepX's
design.

The grouping of all configuration options in Config.defaults.h makes it
easier to see what options are available all in one place, but I have a
nagging desire to move the related #defines into their respective class
header files so keeping related things together.

I mean really are users going to go hunting for Config.defaults.h or just
use the Wiki?

kbahey, 2021-07-22:
I think having all the defaults in one place, and all of them with ifndef, will make it easier to see what gets set if not set in Config.h. 

On the other hand, since this is not supposed to be a user visible (nor editable) file, then it is fair game for the developer to modularize them. 

I am more concerned that Config.h remains the only place where users can edit parameters, and you seem to agree with this. Let us not shock the users with drastic changes (to user editable parts).