hjd1964/OnStepX #21: Compile Error for STEP_WAVE_FORM == SQUARE

michael-ring, 2022-05-18:
main does not compile atm because first parameter for 
StepDirMotor::move(const int16_t stepPin)
is int16_t in header and int8_t in implementation. Assuming that int16_t is correct this fixes the issue:

diff --git a/src/lib/axis/motor/stepDir/StepDir.cpp b/src/lib/axis/motor/stepDir/StepDir.cpp
index f211501..28c86bc 100644
--- a/src/lib/axis/motor/stepDir/StepDir.cpp
+++ b/src/lib/axis/motor/stepDir/StepDir.cpp
@@ -315,7 +315,7 @@ IRAM_ATTR void StepDirMotor::updateMotorDirection() {
 }

 #if STEP_WAVE_FORM == SQUARE
-  IRAM_ATTR void StepDirMotor::move(const int8_t stepPin) {
+  IRAM_ATTR void StepDirMotor::move(const int16_t stepPin) {
     if (direction > DirNone) return;

     if (microstepModeControl == MMC_SLEWING_REQUEST && (motorSteps + backlashSteps) % homeSteps == 0) {
@@ -372,7 +372,7 @@ IRAM_ATTR void StepDirMotor::updateMotorDirection() {
     takeStep = !takeStep;
   }

-  IRAM_ATTR void StepDirMotor::moveFF(const int8_t stepPin) {
+  IRAM_ATTR void StepDirMotor::moveFF(const int16_t stepPin) {
     if (microstepModeControl >= MMC_SLEWING_PAUSE) return;

     if (takeStep) {
@@ -385,7 +385,7 @@ IRAM_ATTR void StepDirMotor::updateMotorDirection() {
     takeStep = !takeStep;
   }

-  IRAM_ATTR void StepDirMotor::moveFR(const int8_t stepPin) {
+  IRAM_ATTR void StepDirMotor::moveFR(const int16_t stepPin) {
     if (microstepModeControl >= MMC_SLEWING_PAUSE) return;

     if (takeStep) {

michael-ring, 2022-05-18:
Thank you for the quick fix!

closed 2022-05-18 by michael-ring

hjd1964, 2022-05-18:
Thanks for the heads-up.