hjd1964/OnStepX #23: What are correct settings for TMC2130 SPI+STEALTHCHOP (analogue TMC2130_QUIET in OnStep)

simonachmueller, 2022-05-28:
I've build a simple ESP32 board (pin compatible with MaxESP3, but only ESP32 and 2 TMC2130 SPI V3 from Bigtreetech installed).

I want to know how to enable StealthChop mode? I tried to set TMC2130S as AXIS1_DRIVER_MODEL and AXIS2_DRIVER_MODEL, but after a restart the tracking does not start and I see following log (DEBUG set to VERBOSE):

> MSG: Setup, HAL initalize
> MSG: Setup, start system service task (rate 10ms priority 7)... success
> MSG: Setup, start input sense polling task (rate 1ms priority 7)... success
> MSG: NV, correct key found
> MSG: Axis1, start monitor task (rate 5000us priority 1)... success
> MSG: Axis1, reverting settings to Config.h defaults
> MSG: Axis1, stepsPerMeasure=220015.79, reverse=OFF
> MSG: Axis1, backlash takeup frequency set to 0.10°/s
> MSG: Axis1, adding any home and/or limit senses
> MSG: StepDir1, pins step=18, dir=0, en=12
> MSG: StepDir1, start task to move motor... success
> WRN: StepDir1, incorrect model for current control, disabling current settings
> MSG: StepDir1, init model TMC2130 legacy u-step mode 16X (goto mode OFF)
> MSG: StepDir1, pins m0=13, m1=14, m2=23, decay=23
> MSG: StepDir1, sequencer homes every 1 step(s)
> MSG: Axis2, start monitor task (rate 5000us priority 1)... success
> MSG: Axis2, reverting settings to Config.h defaults
> MSG: Axis2, stepsPerMeasure=220015.79, reverse=OFF
> MSG: Axis2, backlash takeup frequency set to 0.10°/s
> MSG: Axis2, adding any home and/or limit senses
> MSG: StepDir2, pins step=27, dir=26, en=SHARED
> MSG: StepDir2, start task to move motor... success
> WRN: StepDir2, incorrect model for current control, disabling current settings
> MSG: StepDir2, init model TMC2130 legacy u-step mode 16X (goto mode OFF)
> MSG: StepDir2, pins m0=13, m1=14, m2=5, decay=5
> MSG: StepDir2, sequencer homes every 1 step(s)
> MSG: Mount, site get Latitude/Longitude from NV
> MSG: Mount, site get Date/Time from NV
> MSG: Mount, site start sidereal timer task (rate 10ms priority 0)... success
> MSG: Mount, type GEM
> MSG: Mount, reset at home and in standby
> MSG: Mount, limits start monitor task (rate 100ms priority 2)... success
> MSG: Mount, start guide monitor task (rate 2500us priority 3)... success
> MSG: Mount, library allocated 204 catalog records
> MSG: Mount, set tracking sidereal
> MSG: Mount, status start general status task (1s rate priority 4)... success
> MSG: Axis1, enabled
> MSG: Axis2, enabled
> MSG: Mount, set date/time is unknown so limits are disabled
> MSG: Mount, start tracking monitor task (rate 1000ms priority 6)... success
> MSG: Setup, start command channel A task (priority 5)... success
> MSG: Setup, start command channel B task (priority 5)... success
> MSG: Setup, start command channel BT task (priority 5)... success
> MSG: Setup, start command channel Local task (priority 5)... success

Here is the output when I set TMC2130 (without S)

> MSG: Setup, HAL initalize
> MSG: Setup, start system service task (rate 10ms priority 7)... success
> MSG: Setup, start input sense polling task (rate 1ms priority 7)... success
> MSG: NV, correct key found
> MSG: Axis1, start monitor task (rate 5000us priority 1)... success
> MSG: Axis1, reverting settings to Config.h defaults
> MSG: Axis1, stepsPerMeasure=220015.79, reverse=OFF
> MSG: Axis1, backlash takeup frequency set to 0.10°/s
> MSG: Axis1, adding any home and/or limit senses
> MSG: StepDir1, pins step=18, dir=0, en=12
> MSG: StepDir1, start task to move motor... success
> MSG: StepDir1, init model TMC2130 SPI u-step mode 16X (goto mode OFF)
> MSG: TmcDriver, init RSENSE=0.13
> MSG: SoftSpi, init MOSI=13, SCK=14, CS=23, MISO=4
> MSG: StepDir1, TMC Ihold=350mA, Irun=700mA, Igoto=700mA
> MSG: StepDir1, TMC standstill automatic current calibration
> MSG: StepDir1, sequencer homes every 1 step(s)
> MSG: Axis2, start monitor task (rate 5000us priority 1)... success
> MSG: Axis2, reverting settings to Config.h defaults
> MSG: Axis2, stepsPerMeasure=220015.79, reverse=OFF
> MSG: Axis2, backlash takeup frequency set to 0.10°/s
> MSG: Axis2, adding any home and/or limit senses
> MSG: StepDir2, pins step=27, dir=26, en=SHARED
> MSG: StepDir2, start task to move motor... success
> MSG: StepDir2, init model TMC2130 SPI u-step mode 16X (goto mode OFF)
> MSG: TmcDriver, init RSENSE=0.13
> MSG: SoftSpi, init MOSI=13, SCK=14, CS=5, MISO=4
> MSG: StepDir2, TMC Ihold=350mA, Irun=700mA, Igoto=700mA
> MSG: StepDir2, TMC standstill automatic current calibration
> MSG: StepDir2, sequencer homes every 1 step(s)
> MSG: Mount, site get Latitude/Longitude from NV
> MSG: Mount, site get Date/Time from NV
> MSG: Mount, site start sidereal timer task (rate 10ms priority 0)... success
> MSG: Mount, type GEM
> MSG: Mount, reset at home and in standby
> MSG: Mount, limits start monitor task (rate 100ms priority 2)... success
> MSG: Mount, start guide monitor task (rate 2500us priority 3)... success
> MSG: Mount, library allocated 204 catalog records
> MSG: Mount, set tracking sidereal
> MSG: Mount, status start general status task (1s rate priority 4)... success
> MSG: Axis1, enabled
> MSG: Axis2, enabled
> MSG: Mount, set date/time is unknown so limits are disabled
> MSG: Mount, start tracking monitor task (rate 1000ms priority 6)... success
> MSG: Setup, start command channel A task (priority 5)... success
> MSG: Setup, start command channel B task (priority 5)... success
> MSG: Setup, start command channel BT task (priority 5)... success
> MSG: Setup, start command channel Local task (priority 5)... success

It seems like TMC2130S does not initialize SPI at all, and basically it can't set required mode.

In the source code, I don't see where TMC2130S is implemented, all `if (model == TMC2130)` does not check for TMC2130S

Is the StealthChop supported at all, and if yes, how it should be configured?

hjd1964, 2022-05-28:
You specifically requested the **TMC2130S** model.

The "S" part means **stand alone** i.e. not SPI.

Use the **TMC2130** model.  The default for it is to use stealthChop for tracking and spreadCycle for slews which is 99% of the time the most optimal setting.  Use AXISn_DRIVER_DECAY and _DECAY_GOTO if you want other.

closed 2022-05-28 by hjd1964

hjd1964, 2022-05-28:
Sorry, didn't read all the way through the first time around and you tried **TMC2130** also.  Doesn't change anything, use that.

simonachmueller, 2022-05-28:
I'm not sure it really uses silent mode of the driver, because GoTo and guide movements are quite noisy. My test program (uses https://github.com/teemuatlut/TMCStepper) move the motors definitely almost silent with the same microsteps/currents settings.

Do you think there is any way to figure it out, why the movements are such noisy with OnStepX?

hjd1964, 2022-05-28:
Did you...
Use AXISn_DRIVER_DECAY and _DECAY_GOTO if you want other.

simonachmueller, 2022-05-28:
Yes, just tried as you mentioned it. Did not change anything, still noisy.

Here is my settings
```
/* ---------------------------------------------------------------------------------------------------------------------------------
 * Configuration for OnStepX
 *
 *          For more information on setting OnStep up see http://www.stellarjourney.com/index.php?r=site/equipment_onstep 
 *                      and join the OnStep Groups.io at https://groups.io/g/onstep
 * 
 *           *** Read the compiler warnings and errors, they are there to help guard against invalid configurations ***
 *
 * ---------------------------------------------------------------------------------------------------------------------------------
 * ADJUST THE FOLLOWING TO CONFIGURE YOUR CONTROLLER FEATURES ----------------------------------------------------------------------
 * <-Req'd = always must set, <-Often = usually must set, Option = optional, Adjust = adjust as req'd, Infreq = infrequently changed
*/
//      Parameter Name              Value   Default  Notes                                                                      Hint

// =================================================================================================================================
// CONTROLLER ======================================================================================================================

// PINMAP ---------------------------------------------------------- see https://onstep.groups.io/g/main/wiki/6-Configuration#PINMAP
#define PINMAP                        MaxESP3 //    OFF, Choose from: MksGenL2, MiniPCB2, MaxPCB2, MaxESP3, CNC3, STM32Blue,     <-Req'd
                                          //         MaxSTM3, FYSETC_S6_2, etc.  Other boards and more info. in ~/src/Constants.h

// SERIAL PORT COMMAND CHANNELS ------------------------------------ see https://onstep.groups.io/g/main/wiki/6-Configuration#SERIAL
#define SERIAL_A_BAUD_DEFAULT        9600 //   9600, n. Where n=9600,19200,57600,115200,230400,460800 (common baud rates.)    Infreq
#define SERIAL_B_BAUD_DEFAULT        9600 //   9600, n. Baud rate as above. See (src/pinmaps/) for Serial port assignments.   Option
#define SERIAL_B_ESP_FLASHING         OFF //    OFF, ON Upload ESP8266 WiFi firmware through SERIAL_B with :ESPFLASH# cmd.    Option
#define SERIAL_C_BAUD_DEFAULT         OFF //    OFF, n. Baud rate as above. See (src/pinmaps/) for Serial port assignments.   Infreq
#define SERIAL_D_BAUD_DEFAULT         OFF //    OFF, n. Baud rate as above. See (src/pinmaps/) for Serial port assignments.   Infreq
#define SERIAL_E_BAUD_DEFAULT         OFF //    OFF, n. Baud rate as above. See (src/pinmaps/) for Serial port assignments.   Infreq

// USER FEEDBACK -------------------------------------------- see https://onstep.groups.io/g/main/wiki/6-Configuration#USER_FEEDBACK
#define STATUS_LED                    OFF //    OFF, Steady illumination if no error, blinks w/error code otherwise.          Option

// RETICLE CONTROL ------------------------------------------------ see https://onstep.groups.io/g/main/wiki/6-Configuration#RETICLE
#define RETICLE_LED_DEFAULT           OFF //    OFF, n. Where n=0..255 (0..100%) activates feature sets default brightness.   Option
#define RETICLE_LED_MEMORY            OFF //    OFF, ON Remember reticle brightness across power cycles.                      Option

// WEATHER SENSOR ------------------------------------------------- see https://onstep.groups.io/g/main/wiki/6-Configuration#SENSORS
#define WEATHER                       OFF //    OFF, BME280 (I2C 0x77,) BME280_0x76, BME280_SPI (see pinmap for CS.)          Option
                                          //         BMP280 (I2C 0x77,) BMP280_0x76, BMP280_SPI (see pinmap for CS.)
                                          //         BME280 or BMP280 for temperature, pressure.  BME280 for humidity also.

// MOTION ---------------------------------------------------------- see https://onstep.groups.io/g/main/wiki/6-Configuration#MOTION
#define STEP_WAVE_FORM             SQUARE // SQUARE, PULSE Step signal wave form faster rates. SQUARE best signal integrity.  Adjust
                                          //         Applies to all axes.

// =================================================================================================================================
// MOUNT ===========================================================================================================================

// Driver models (Step/Dir and Servo) many have specific requirements so be sure to follow the link below to help learn about these.
// Typically: A4988, DRV8825, LV8729, S109, TMC2130, TMC5160, TMC2209S, TMC2209U, TMC2226S, TMC2226U
// Others: SERVO_II and SERVO_PD have settings in Extended.config.h

// AXIS1 RA/AZM -------------------------------- see https://onstep.groups.io/g/main/wiki/Configuration---Rotator-and-Focusers#AXIS1
#define AXIS1_DRIVER_MODEL            TMC2130 //    OFF, Enter motor driver model (above) in both axes to activate the mount.    <-Often

// If runtime axis settings are enabled changes in the section below may be ignored unless you reset to defaults:
// \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ \/ 
#define AXIS1_STEPS_PER_DEGREE      3840 //  12800, n. Number of steps per degree:                                          <-Req'd
[... cut ...]
#define FEATURE8_PIN                  OFF //    OFF, AUX for auxiliary pin, n. Where n is the pin#.                           Adjust
#define FEATURE8_VALUE_DEFAULT        OFF //    OFF, ON, n. Where n=0..255 for ANALOG_OUT purpose.                            Adjust
#define FEATURE8_ON_STATE            HIGH //   HIGH, LOW to invert so "ON" is 0V and "OFF" is Vcc (3.3V usually.)             Adjust

// ---------------------------------------------------------------------------------------------------------------------------------
#define FileVersionConfig 5
#include "Extended.config.h"
```

simonachmueller, 2022-05-28:
I also played a bit by setting AXIS1_DRIVER_MICROSTEPS_GOTO, but no help.

hjd1964, 2022-05-28:
The quiet mode is STEALTHCHOP, use it for _GOTO also if that's what you really want.

simonachmueller, 2022-05-28:
Yes, STEALTHCHOP for _GOTO did the magic I want, thank you so much!
Btw, why is that not recommended/not default?

hjd1964, 2022-05-28:
SPREADCYCLE achieves higher motor RPM's.