hjd1964/OnStepX #24: Strange behaviour when connected to ASCOM driver

simonachmueller, 2022-05-29:
When OnStepX is connected to the ASCOM driver, the RA motor only slews west, regardless of the command (slew east/west).
Steps to reproduce:
1. Start OnStep
2. Connect the mobile app using Bluetooth
3. Go to Guide/Focus in the app, move west/east: motor rotates in different directions
4. Connect latest OnStep ASCOM driver
5. Go to Guide/Focus in the app, move west/east: motor rotates always in same direction
6. Try slewing from the ASCOM Device Hub: same behavior
7. Disconnect ASCOM drivers
8. Go to Guide/Focus in the app, move west/east: motor rotates in different directions

Can't be reproduced with Stellarium when using native LX200 communication. 
Can be reproduced with Stellarium when using ASCOM communication.

simonachmueller, 2022-06-02:
Hi @hjd1964, did you have any chance to look into the issue or could you point me to the source code place to look into it? I guess Goto.cpp is responsible for that part, is that correct? The code is quite complicated here.

simonachmueller, 2022-06-11:
Here is my configuration [config.zip](https://github.com/hjd1964/OnStepX/files/8883668/config.zip)

simonachmueller, 2022-06-11:
As per your message https://onstep.groups.io/g/main/message/44580, I downloaded latest sources, but the behaviour is the same.
I'm using ASCOM driver ver 3.11, OnStepX 10.07i

simonachmueller, 2022-06-20:
Can't reproduce it when connected using IP/network interface 🤔

hjd1964, 2022-06-24:
Think something was mentioned about the ASCOM connection being over Serial where in my first attempt to reproduce this I used WiFi/IP.  I tried the test again with serial and everything worked fine... still no issues here.

simonachmueller, 2022-06-24:
@hjd1964 Do you use version of the driver you published on your webpage or some dev version? Another difference could be that my build does not have any RTC or GPS time source. Were you able to use my config I uploaded in a previous comment to test?

hjd1964, 2022-06-24:
- Do you use version of the driver you published on your webpage or some dev version?

Same as my website.

- Another difference could be that my build does not have any RTC or GPS time source.

Same, no time source.

- Were you able to use my config I uploaded in a previous comment to test?

Yes

closed 2022-11-16 by hjd1964

shimpa1, 2023-10-27:
Hate to bring an old thread out of history but how did you resolve this?

simonachmueller, 2023-11-08:
@shimpa1 I can't remember, but now my OnStepX works fine. Do you have similar issue?

shimpa1, 2023-11-08:
Yes I had a similar issue.
I haven't narrowed it down exactly however I found out that if there's more than one connection to the mount (like a Bluetooth and a USB connection simultaneously) then it might start acting like that.
After disconnecting the USB from the mini PC and using the OnStep Android app to slew it back to home position manually then resetting and then restarting the mount again it is now back to normal.
Another thing that helped was unchecking the serial error correction and DTR control in ASCOM driver.
hope it helps,

CyberianIce, 2024-04-03:
I think I may experienced this last night on star party. After first sync command, RA only goes west, and only when ASCOM is connected via USB. That DTR control thing may have something to do with this issue because when disabled, i think it works well.

shimpa1, 2024-04-04:
Correct, the serial port data flow settings must be disabled and the everything works as expected.

CyberianIce, 2024-04-04:
Yes, new ASCOM driver confused me because after update all settings are set to defaults, and DTR control is now displaced in dropdown menu with baud rates so I didn't see there are boud rates without DTR control on the bottom of it. I used to set baud rates via ASCOM Profile Explorer before it was possible to do so with OnStep driver dialog.