hjd1964/OnStepX #58: Return of the command :SCMM/DD/YY#

mvackel, 2024-04-17:
The protocol implementation of the LX200 command :SCMM/DD/YY# is returning "1#"  (when the date is valid). This is causing a long delay to the Skysafari connection as commented in the Wiki, when configured to set time/location.

The original LX200 returns exactly the following string after receiving this command (the number of spaces matter) (the "s are not part or the return):

"1Updating        planetary data. #                                #"

(As a side note, actually it gives two returns. First returns "1Updating        planetary data. #" and after about 4 seconds, 
returns "                                #" )

I've tested this using the following code in the file "src/telescope/mount/site/Site.commands.cpp":

---- line 138:
  if (command[0] == 'S') {
    // :SC[MM/DD/YY]# or :SC[MM/DD/YYYY]#
    //            Change local standard date
    //            Return: 0 on failure, 1 on success
    if (command[1] == 'C') {
      GregorianDate local = calendars.julianToGregorian(UT1ToLocal(getDateTime()));
      if (strToDate(parameter, &local)) {
        dateIsReady = true;
        setDateTime(localToUT1(calendars.gregorianToJulian(local)));        
        updateTLS();
        // --- modification:   (the last '#' is added after this funcion)
	sprintf(reply,"%s", "1Updating        planetary data. #                                ");
        *numericReply = false;
        // --- End modification
      } else *commandError = CE_PARAM_FORM;
    } else                               
.........

This modification makes Skysafari connection happen almost immediately, solving the problem. 
I tested this modification using Skysafari Pro version 7, setting telescope to "LX200 Classic", LXD650 and LXD55. 

Using Skysafary Pro version 6 and telescope "LX200 Classic", there is still a delay of about 30 seconds, but using LXD650 or LXD55 works normally and the connection happens immediately.

vzr314, 2024-04-17:
Using LXD will open another rabbit hole you don't want to go through. Just set TLS with OnStep app and use LX200 Classic, works the best with both SkySafari 6 and 7, via Bluetooth and WiFi.

mvackel, 2024-04-19:
Well, I'm using iOS and setting Time/Location using Skysafari is MUCH easier - and automatic. Besides I use to observe from multiple locations, so this feature is very convenient.
About using LXD650 or LXD55, I haven't had problems yet, as the protocol is the same. Specially LXD650 that uses the same firmware and the electronics is almost exactly the same of the LX200 Classic (I have both here and fixed the electronic boards many times). Would you, please, comment about the problems you noticed?
Anyway if you could merge these modifications to the project, the problem would be solved conveniently.

hjd1964, 2024-09-17:
My feeling is this change would likely break something else, so it's staying as-is.