hjd1964/OnStepX #67: compile Teensy 4.1 error

ezioall59, 2024-11-24:
hi
compiling with Arduino 1.8.9 I get this very strange error.
the missing file is present
why this error??

Più di una libreria trovata per "TimeLib.h"
In file included from /home/ezio/Arduino/OnStepX/src/plugins/sample/Sample.cpp:3:
Usata: /home/ezio/Arduino/libraries/Time-1.6.1
/home/ezio/Arduino/OnStepX/src/plugins/sample/Sample.h:4:10: fatal error: ..\..\lib\commands\CommandErrors.h: No such file or directory
    4 | #include "..\..\lib\commands\CommandErrors.h"
Non usata: /home/ezio/arduino-1.8.19/hardware/teensy/avr/libraries/Time
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Errore durante la compilazione per la scheda Teensy 4.1.

papadako, 2024-11-26:
Can you please check the following?

https://github.com/hjd1964/OnStepX/pull/69

ezioall59, 2024-11-27:
hi
i tried from windows 10
first from a folder already unzipped and it doesn't work, while unzipping directly from window works
so for a strange reason from Fedora 41 it doesn't work
so i close this problem
the cause is not in OnStepX

closed 2024-11-27 by hjd1964