hjd1964/OnStepX #76: Both ST4_INTERFACE and ST4_HAND_CONTROL ON Issue

testezt, 2025-02-23:
Even though there is no SHC activated, ST4 guide rate is set as guide.settings.axis1RateSelect and guide.settings.axis2RateSelect. 
I think it must check if there is a SHC attached not the ST4_HAND_CONTROL value. This code should be modified.
bool pulseGuiding = GUIDE_SEPARATE_PULSE_RATE == ON && ST4_HAND_CONTROL != ON;
-> bool pulseGuiding = GUIDE_SEPARATE_PULSE_RATE == ON && !shcActive;

hjd1964, 2025-02-23:
You don't understand the code.

pulseGuiding selects between using pulse guide rates or guide rates for the ST4 port when a traditional ST4 device is attached.

That selection is dictated by the GUIDE_SEPARATE_PULSE_RATE and ST4_HAND_CONTROL settings.  An SHC being active on the port has nothing to do with it.

closed 2025-02-23 by hjd1964

testezt, 2025-02-24:
OnStep legacy ST4 port works strangely when both ST4_HAND_CONTROL ON and GUIDE_SEPARATE_PULSE_RATE ON without0 SHC.
bool pulseGuiding = GUIDE_SEPARATE_PULSE_RATE == ON && ST4_HAND_CONTROL != ON;
This code gives pulseGuiding=false and it makes use of axis1RateSelect instead of pulseRateSelect even though I don't have SHC.
guide.startAxis1(st4GuideActionAxis1, pulseGuiding ? guide.settings.pulseRateSelect : guide.settings.axis1RateSelect, GUIDE_TIME_LIMIT*1000);
Is this correct operation even though I don't have SHC?